suchmememanyskill/CYD-Klipper #138: Support for JC4827W543N/R/C

samborambo, 2025-01-09:
Hi there,

I have a JC4827W543C from Sunton on the way as it has a few extras over the 'S043. I'd like to use CYD_Klipper but I can't see support in the code.

Touch sensor is the same as the 'S043. It's qSPI with NV3041A instead of par565. Support requires LVGL 8.4.0.

There's a BSP and quite a few repos with support around. Like here https://github.com/profi-max/JC4827W543_4.3inch_ESP32S3_board

Happy to help with the PR if I knew where to start!

suchmememanyskill, 2025-01-09:
Why is lvgl 8.4.0 required?

You could look into writing a driver here: https://github.com/suchmememanyskill/CYD-Klipper/tree/master/CYD-Klipper/src/core/device

samborambo, 2025-01-14:
Thanks! I'll have a crack at it.

Not sure about the LVGL version. I noticed it in the repo I linked in the ticket. You probably understand LVGL better than me. I'll try with this repo's version.

closed 2025-01-14 by samborambo