suchmememanyskill/CYD-Klipper #140: v2.0.0 - FPS counter and CPU load indication

dmit2k, 2025-01-09:
Everything works perfect so far!

But IMHO there should be a configuration option to disable FPS counter and CPU load indication on the screen... :)

suchmememanyskill, 2025-01-10:
Did you flash via web or via manual compile?

Debug builds always have an fps counter. Release builds don't

dmit2k, 2025-01-10:
> Did you flash via web or via manual compile?

Flashed manually from the Git Master branch, that makes me think it should be Release build.

Flashing via Web seems to finish OK, but puts device into dead loop (same as before).

suchmememanyskill, 2025-01-11:
@dmit2k i made an alt version with your suggested changes. This one does not work?

For building manually: you have to be on the exact same commit as the release was made

dmit2k, 2025-01-11:
@suchmememanyskill 

Web flashing with both board versions goes into dead loop:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
...

Flashed successfully manually with Platformio

suchmememanyskill, 2025-01-28:
Made a seperate issue for webflash/compile not working for S3 images.
Closing as the issue itself is cause user is running a debug build

closed 2025-01-28 by suchmememanyskill