suchmememanyskill/CYD-Klipper #142: serial/usb connection

KennyVB, 2025-01-10:
LOVE THE NEW UPDATE

how ever, the ESP is not visable on the system as per say /dev/ttyUSB noir in by-device
so i can't get it to connect, just updated the device from webinstaller, and made the host install, no error in systemctl --user status so guessing it works, but dont think it can find the esp, as i cant find it on the subsystem. using btt pi1.2 but it's just linux

suchmememanyskill, 2025-01-10:
Any info in dmesg?

KennyVB, 2025-01-11:
none about it finds the esp. only my webcam tho

suchmememanyskill, 2025-01-11:
I'm unsure how i can help you here. Maybe the drivers shipped in your flavor of linux don't include usb serial? Maybe you're not using a data usb cable?

KennyVB, 2025-01-18:
linux works, usb devices work, somehow, the cyd doesnt show up. can find multiple other devices in usb. so what gives. not sure whats in the firmware your putting out tho

KennyVB, 2025-01-18:
using the usb-c btw

KennyVB, 2025-01-18:
[image]

suchmememanyskill, 2025-01-18:
Configuration issue from your side, i'd say. The user you install it under does need to always be running

KennyVB, 2025-01-19:
well you made it installed in user space, which for many isent configured.
dont understand why it has to run there, just a service file

suchmememanyskill, 2025-01-19:
Not really.

Iirc in mainsailos you log in as the same user as that runs Klipper.

Either way, would you mind making a pr for this, to run it as root? (Remove the user bit)

If you do, please target the dev branch.

suchmememanyskill, 2025-01-28:
@KennyVB I just added a fix in the dev branch to install the service as root (uninstall first, then just sudo ./install.sh. You will be asked if you want to install under the root user). Can you give this a try?

KennyVB, 2025-02-02:
wonderfull work champ! thank u!