suchmememanyskill/CYD-Klipper #143: Works in ttyUSB0 not ttyS1

mechano, 2025-01-11:
Hi there, I'm trying the new version.

My CYD 2.8" ESP32-2432S028 works with WiFi, and the new serial version only on ttyUSB0. 

But can't connect to the ttyS1 port that was of the previoud Nextion/TJC monitor. 

The  CYD has both an USB port  and a port with a jst1.25 4 way with label 5v, RX, TX, GND. 

I've to say that I've full control of my printer with a fresh Armbian based on Ubuntu Noble on a SKIPR mainboard. ttyS0 is the THR connected by UART. And I saw previous display was a ttyS1. There's also ttyS2 but I've tried both.

Cabling is correct and I've tried both directions of TX and RX.

[image]

suchmememanyskill, 2025-01-11:
Some other user reported that the cyd 2.8" contains a design defect that leaves TX disconnected https://github.com/suchmememanyskill/CYD-Klipper/issues/100#issuecomment-2508731774

zefir-o, 2025-01-13:
See comment 
https://randomnerdtutorials.com/esp32-cheap-yellow-display-cyd-pinout-esp32-2432s028r/#comment-961977

zefir-o, 2025-01-13:
To be able to use that port, I manually recompiled the library and swapped TX/RX in the initialization of the serial.

mechano, 2025-01-18:
It seems also the 3.2" capacitive doesn't work with ttyS1

suchmememanyskill, 2025-01-28:
Closing as it's not an issue with CYD-Klipper

closed 2025-01-28 by suchmememanyskill