suchmememanyskill/CYD-Klipper #144: Connection thru ttyUSB0 no macro available

mechano, 2025-01-11:
I've set some useful macros with the Description: CYD_SCREEN_MACRO
and can see connecting by Wifi. 

Now I'm testing USB connection to the same machine. Everything works but it says there're no macros and add the Description: CYD_SCREEN_MACRO to display them.

but the description already exists.

suchmememanyskill, 2025-01-11:
Would you mind sharing some logs of the host side serial program? I did implement this, that's odd

mechano, 2025-01-11:
Sorry, where does it store logs?

suchmememanyskill, 2025-01-11:
It outputs it to stdout.

Either way, set up my own environment again. I think i found the issue.
When you have no power devices, the request for power devices fails. This causes some data to be left in the receiving buffer, that blocks the next request from working properly. See commit https://github.com/suchmememanyskill/CYD-Klipper/commit/fdfa2d014ffa87e1f45e88b41fb7a16be5132261

Could you test this commit?

Build: https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/12725738302 (it's currently still building, will take roughly 15 mins to complete)
Instructions: https://github.com/suchmememanyskill/CYD-Klipper/blob/master/Manual_Install.md#flashing-a-build-via-the-commandline

mechano, 2025-01-11:
How do I flash manually, I've used web based interface till now.

mechano, 2025-01-11:
My screen bricked trying to apply this update. 
At first the upload hanged at 40%, after finished but the content was not erased, another time stopped at 80%. 
Another it finished but since then the screen is black and doesn't switch on. 

I think it's broken. 

I'll take a couple of 3.2" capacitive.

suchmememanyskill, 2025-01-11:
Unstable usb connection? (User error? The builds you have there are the same kind of builds that get released)

I assume it starts working again when you flash the release (web) version back?

mechano, 2025-01-11:
I tried the web version and also another ESP32 project around the web.
Nothing, the display doesn't switch on.
I think has broken.

I've ordered a couple of 3.2" capacitive with the p.n. listed in your
project's page.

I like your project and it's growing very well. With serial support I
think soon we will be able to change any stock Nextion/TJC crap from many
printers.

suchmememanyskill, 2025-01-11:
That is incredibly odd....

Even when i mess up pin definitions, i can just reflash and make it work. I guess you flashed something that had a specific enough config to break something?

(Wrong screen maybe? The firmware output folder just has codenames)

Note: i don't provide the links for where to buy, you likely got the link from the guy who makes board definitions (esp32 smartdisplay)

suchmememanyskill, 2025-01-11:
[image]
[image]

Cannot reproduce

mechano, 2025-01-11:
Yes I've the same messages, but the mine has the screen black, only a red led on the back in on. The backlight of the screen in off.

mechano, 2025-01-12:
The command esptool write_flash 0x0 and new flash of CYD from the web based installer unbricked my screen. 
Now the LCD backlight is already on and everything works.

mechano, 2025-02-01:
Still same problem with 2.1.0 from web interface. It recognize the display with BOOT putton pressed, erase and install, but it results with a blank display. No backlight, nothing. 

3.2" capacitive ESP32-2432S032

suchmememanyskill, 2025-02-01:
Logs?

Again, cannot reproduce. Installs fine on my 3.2" Capacitive screen. Are there different revisions of the screen? Installed via edge on windows 11.

mechano, 2025-02-01:
I use Chrome on Ubuntu Linux 24.10