suchmememanyskill/CYD-Klipper #146: If M106 is overridden no part fan speed control in CYD

mechano, 2025-01-12:
Hi there!

Discovered just now. 

Into Orca Slicer Github page they explain how to override the M106 to control part, auxiliary and filter fans.

https://github.com/SoftFever/OrcaSlicer/wiki/air-filtration

But if I do it the CYD looses control of the part fan.

mechano, 2025-01-13:
It's not a problem of CYD, it can't support slicer's specific controls.

closed 2025-01-13 by mechano

suchmememanyskill, 2025-01-13:
How so? Cyd-klipper generates M106 commands? (https://www.klipper3d.org/G-Codes.html)

If you open either serial console for wifi, or get the logs of serial server, it gives you exactly what goes over the wire.

Note having a printer in cyd-klipper with type klipper serial configured will disable the serial console

suchmememanyskill, 2025-01-13:
Orcaslicer's wiki seems to implement M106 in a backwards compatible way with optional args

mechano, 2025-01-13:
Found the solution, at almost for my printer with enclosure and both auxiliary and filter fans.

After definitions, I override the M106, but leave also the standard behaviour to let CYD still control the part fan.

```
# Fan pin definitions
[fan]
pin: MKS_THR:gpio3

[fan_generic auxiliary_fan]
pin: PC9
max_power: 1.0
shutdown_speed: 0.0
cycle_time: 0.0005 #2 kHz PWM signal
hardware_pwm: False
kick_start_time: 0

[fan_generic filter_fan]
pin: PC12
max_power: 1.0
shutdown_speed: 0.0
cycle_time: 0.0005 #2 kHz PWM signal
hardware_pwm: False
kick_start_time: 0

# Used by Orca Slicer to control also Auxiliary and Filter fan
[gcode_macro M106]
rename_existing: M106.1
gcode:
  {% set fan = 'fan' + (params.P|int if params.P is defined else 0)|string %}
  {% set speed = (params.S|float / 255 if params.S is defined else 1.0) %}
  {% if fan == 'fan2' %}
    SET_FAN_SPEED FAN=auxiliary_fan SPEED={speed}
  {% endif %}
  {% if fan == 'fan3' %}
    SET_FAN_SPEED FAN=filter_fan SPEED={speed}
  {% endif %}
  {% if params.S is defined and fan == 'fan0' %}
    {% set speed = params.S|string %}
    M106.1 S{speed}
  {% endif %}
```