suchmememanyskill/CYD-Klipper #147: 2.00 platformio.ini  esp32-4827S043R

DragonSlayerSE, 2025-01-13:
there is an spelling error on row 87
should be 
[env:esp32-4827S043R-SD]
board = esp32-4827S043R-smartdisplay

(there is a C instead of an R)

suchmememanyskill, 2025-01-13:
The c indicates capacitive, this is not an error

DragonSlayerSE, 2025-01-14:
i am aware of that.. the ESP-4827S043R enters an crashloop if you compile it with the Capasitive touch drivers that platformio.ino refers to on row 87, just change the C to an R and it will work.. but there is still problem with the calibration of the touch

suchmememanyskill, 2025-01-14:
I never finished the implementation, its not released.....

https://github.com/suchmememanyskill/CYD-Klipper/blob/master/CYD-Klipper%2Fboards%2Fesp32-4827S043R-smartdisplay.json#L91 you likely need to change the touch rotation. Feel free to send a pr with the correct values

closed 2025-01-14 by suchmememanyskill

DragonSlayerSE, 2025-01-14:
i am new to github but i try to make a RP