suchmememanyskill/CYD-Klipper #156: SC01 Plus

jonboy86, 2025-01-21:
Is there any chance that the SC01 is able to be used ? these are cheap on ali express and the like

suchmememanyskill, 2025-01-23:
I can't add support for screens i do not have. If you're happy to try, please send a pull request of an implementation

jonboy86, 2025-01-23:
ill give setting up  a dev enviroment a go

spenc302, 2025-04-17:
Any progress made? I'm trying to do so myself currently and am running into bootloop issues due to hardware settings I'm trying to get right currently.
Will gladly share my cpp Json and updated ini files of you would like them to try and get it going on your end, what variant do you have? I'm not sure how much pins varey but I know ram can greatly. This is set for N8R2 but again I'm not sure if it a ram settings or pins currently