suchmememanyskill/CYD-Klipper #160: Control CYD LED

SartorialGrunt0, 2025-01-26:
Some CYD ESP32 displays have an onboard RGB LED such as ESP32-3248S035C. I would be cool to be able to turn it on and change its color if possible.

suchmememanyskill, 2025-01-26:
What would this led do? Backend code for this is partially implemented already, i'm just unsure what to do with it (so for the screens its implemented on, it turns the led off)

SartorialGrunt0, 2025-01-26:
> What would this led do? Backend code for this is partially implemented already, i'm just unsure what to do with it (so for the screens its implemented on, it turns the led off)

I'd honestly be happy with just getting to change the color in the UI, preferably with a RGB 0-255 value, so it can match my other NeoPixels.

Some other ideas:
- CYD connection status
- Controllable with a klipper macro 
- Indicate Print Status
- Match the System Theme Color
- Status of button press action received.