suchmememanyskill/CYD-Klipper #161: No way to connect anything

retikulumx, 2025-01-26:
Hey.
Neither I'm able to add the Octoprint API key, which I had to type manually, checked it multiple times, but "key incorrect".
Nor can I type the Bambu Serial number since the keyboard doesn't even provide a P or S for Bambus printer's serial number. Is this a joke or a hidden secret somewhere?

suchmememanyskill, 2025-01-26:
Why was this issue closed? Is this still an issue?

> octoprint api key

Should probably find a better way of letting the user enter this, but its absurd octoprint requires a 48 char long api key. You can use the serial console *before* going into the octoprint config menu, type `key <octoprintkey>`, then enter the menu.

> keyboard serial for bambu

I only have an a1 mini to test on. Do serials have different structures between printers?

Note i'm not talking about the device id (xxx-xxx-xxx), but the printer SN

retikulumx, 2025-01-26:
Hi @suchmememanyskill 
wow thanks so much for the very quick response!

I'm gonna try this serial console variant, thanks for this. Yep, this long API key is a pain!

And yes, the different printers' SN have different structures. The SN of the P1S starts with "01P" and the SN of the P1P starts with "01S" (yes, indeed): https://wiki.bambulab.com/en/general/find-sn

suchmememanyskill, 2025-01-26:
Oh, that's awful. I'll change the keyboard type next release.

As a workaround, you can use the same trick as the octoprint key, using the serial console before entering the menu

retikulumx, 2025-01-26:
Thanks I'm gonna try this.
And no idea why the issue was closed initially. I might have closed it accidentally.

retikulumx, 2025-01-27:
Ok, I tried the options with key via console. Works great! :)
Now I'm gonna play around a bit. I might use it as a replacement for the 2 color display of my old Ender 5 (with Klipper/Octoprint).
It also works for the Bambu now by skipping the keyboard. But there's an app for my Bambu, so.... ^^

closed 2025-02-01 by suchmememanyskill