suchmememanyskill/CYD-Klipper #162: CYD-klipper-serial not starting at boot

SartorialGrunt0, 2025-01-26:
I've noticed that the cyd-klipper-serial service is refusing to start at bootup. If I manually start the service with a "systemctl --user start cyd-Klipper-serial" it comes up fine.

I tried running the command "systemctl --user enabled cyd-klipper-serial" and verified that it's enabled with an "is-enabled" command. This did not resolve it. 

Hardware/OS: Pi 3B+, latest rasbian lite 32bit

Let me know if there is any more details I can provide. I suspect this may be unique to my install but I'm unsure. I'm guessing another service is getting in the way or the "WantedBy=multi-user.target" isnt calling the service for some reason.

suchmememanyskill, 2025-01-26:
Is your user being used on startup? As in, is a session created under your user during boot?

I wonder if i should use a root version (what does --user do when you're the root user? Does it act like you just invoked it without user?)

SartorialGrunt0, 2025-01-26:
> Is your user being used on startup? As in, is a session created under your user during boot?
> 
> I wonder if i should use a root version (what does --user do when you're the root user? Does it act like you just invoked it without user?)

Doesn't klipper normally run in user space? So I would think a session is being created.

Running a systemctl --user start gives the following response:

Failed to connect to bus: $DBUS_SESSION_BUS_ADDRESS and $XDG_RUNTIME_DIR not defined (consider using --machine=<user>@.host --user to connect to bus of other user)

So yeah looks like it's complaining you aren't a user.

suchmememanyskill, 2025-01-28:
@SartorialGrunt0 I just added a fix in the dev branch to install the service as root (uninstall first, then just sudo ./install.sh. You will be asked if you want to install under the root user). Can you give this a try?

SartorialGrunt0, 2025-01-29:
> @SartorialGrunt0 I just added a fix in the dev branch to install the service as root (uninstall first, then just sudo ./install.sh. You will be asked if you want to install under the root user). Can you give this a try?

Installing from root with your dev build does indeed resolve the issue!

closed 2025-02-01 by suchmememanyskill