suchmememanyskill/CYD-Klipper #168: Not proper colors appearing

TheSpookys, 2025-02-22:
After flashing the software to the display the colors seem to be inverted and I had to enable invert colors to have the proper colors appearing.

This is when it is disabled
[image]

This is with it enabled
[image]

suchmememanyskill, 2025-02-22:
This is not an issue with the software; The screen itself contains a bug where this is needed. The other variants of the 2.8" resistive screen do not have this issue. This is why the invert colors toggle exists in the first place.

closed 2025-02-22 by suchmememanyskill

TheSpookys, 2025-02-22:
Ohhh that makes much more sense!