suchmememanyskill/CYD-Klipper #169: [SUGGESTION] Add a console menu and circular progressbar

TheSpookys, 2025-02-23:
Could you please add a console menu and update the progress bar to a circular one or just add an option for that :)

Thanks!

suchmememanyskill, 2025-02-23:
For the console,

Wouldn't you much rather pull out a phone or pc to type in commands? I don't think the tiny screen makes for a great typing experience. If you frequently need a command, make it a macro, that makes it a button in the macros menu.

For the circular progress bar,
I don't think i have the screen space to pull that off

TheSpookys, 2025-02-23:
For the console I agree with you now that I have bought about it, however the circular progress bar is possible as I had seen a video about it

https://youtube.com/shorts/EcBUoENDC6E?si=PigmaQqB04azkATU

TheSpookys, 2025-02-23:
Whoops I misclicked and closed it lol

suchmememanyskill, 2025-02-23:
That video mentions there is a lot of 'wasted' space. But turn on showing emergency stop and extra info to all, and almost the entire screen is filled

TheSpookys, 2025-02-23:
Ohh yah I just realised that, thanks for the reply tho!

closed 2025-02-23 by TheSpookys