suchmememanyskill/CYD-Klipper #171: Cannot get past calibration screen on ESP32-2432S028 Resistive

rauldelga, 2025-03-03:
I connect the screen to my PC, use chrome, and install the new firmware onto the screen.

When it turns on, the calibration screen does not work. I hold the boot button for 8 seconds and it just resets. Sometimes it'll move the crosshair to the lower right after a boot, only if i press the button at the same time, but i can never get pass the second crosshair no matter how many times i hold the boot button.

suchmememanyskill, 2025-03-03:
Does other software that require the touch functionality break? (As in, is this a software issue on my side or a hardware issue?)

I think XTouch has a similar calibration routine

rauldelga, 2025-03-03:
I'll try to download Xtouch. The touch did work on the stock OS.

I did manage to get past the screen by typing touch 1 1 1 1 in console but the screen did not work. As a hail marry, what would those numbers typically be?

suchmememanyskill, 2025-03-03:
I don't know off the top of my head what those values are supposed to be, i'll check when i'm home.

It's incredibly odd that touch seems to work that inconsistently for you. Can you pass calibration if you tap in the same spot twice?

https://xperiments.in/xtouch-bin/ is the installer website for xtouch.

suchmememanyskill, 2025-03-03:
My calibration values, i doubt they'll be useful to you but just in case they are: Screen coefficients: x_screen = 0.088915 * x_touch + -23.254299;  y_screen = 0.067901 * y_touch + -16.888887

rauldelga, 2025-03-04:
ok, so good news and bad news. installing xtouch did not work. neither did re-installing cyd-klipper

However if i typed into console "touch 0.088915 -23.254299 0.067901 -16.888887" "ssid name pass" "id v0.local" "reset"

It got me to the host ip screen and so far it seems like touch is working. I haven't messed with it much, but i'll update if it stops working. For now, just assume that giving me the calibration inputs seems to have gotten me past the calibration screen.