suchmememanyskill/CYD-Klipper #177: Macro with Octoprint

Dh0mp5eur, 2025-03-24:
Hello,

Are macros with Octoprint supported?

If so, how can I add new macros to my device?

I added macros in Gcode Macro menu on Octoprint but they don't appear in the interface.

Thanks a lot!

suchmememanyskill, 2025-03-24:
Hi, is this a base octoprint feature or a plugin?

I don't currently support this, no

Dh0mp5eur, 2025-03-24:
Hi

This is the Gcode Macros plugin.

https://github.com/cp2004/OctoPrint-GcodeMacros

Dh0mp5eur, 2025-03-24:
I understand.

But I already have 2 functional macros on my device that I didn't create (Auto-Level & Disconnect printer).
Are they embedded in the firmware?

Is it possible to add more?

suchmememanyskill, 2025-03-24:
Yes, they are embedded, and no, it's not possible to add more (well, unless you want to go spelunking in the codebase)