suchmememanyskill/CYD-Klipper #181: Display blinks endlessly when using Serial connection

spammchik, 2025-04-21:
Hi. I bought a display here https://aliexpress.com/item/1005006625303218.html 
It's 4.3 inches. 800*480 - IPS Capacitive touch
Аnd installed the ESP32-8048S043 Alt (4.3" 800x480 Capacitive) firmware via the website https://suchmememanyskill.github.io/CYD-Klipper/
The CYD-Klipper firmware works with this display. Now I am assembling all the electronics together and have some problems.

When connecting the display via a USB cable (Serial connection) to a device with Klipper, the display starts blinking endlessly and it is impossible to work on it. The connection is made and you can even see the list of files.

I also noticed that the display blinks even if the display is simply connected to USB charging (without data), and at the same time you try to start a Serial connection on the monitor.

What could be the reason for the blinking when using a Serial connection?
There is no blinking when connecting WiFi, everything works fine.

I posted a video of the blinking display here https://www.mediafire.com/file/sbobchcjlt3cvfv/VID_20250421.mp4/file

Many thanks to the author of the CYD-Klipper project for the work. I ask everyone who uses CYD-Klipper to support the author financially at https://ko-fi.com/suchmememanyskill
Such projects should be supported.

suchmememanyskill, 2025-04-21:
> and at the same time you try to start a Serial connection on the monitor.

Does this only happen when you start a serial connection from the display, or does this happen the other way around too? (Start from pc?)

This seems super odd, i don't have a 4.3" display so i cannot try to reproduce this either. This at least doesn't happen on my other displays. Does it do this 'flicker' in other spots too? It sounds odd to be timing related, as refreshes may not happen in a specific timeframe

spammchik, 2025-04-21:
Oh no, I don't connect the display to the computer (PC). By monitor I mean the CYD display itself.
And yes, the problem occurs when starting the Serial connection from the CYD display.
There is no display blinking in other places.
> Maybe a misconfigured pin somewhere?

I use a regular USB-C cable and haven't configured it in any way.